repetier/Repetier-Server #73: Feature Request: Add option to recalculate build times

miked63017, 2015-08-12:
If you change settings around in the printer and/or EEPROM the build time remains the same unless you delete and re-upload the files. It would be nice to have either a button somewhere to re-calculate, or to just have it happen automatically when you change EEPROM or printer settings. Personally I think the button would be better because then you could do one piece at a time instead of waiting for your whole library to re-calculate, but either way would be nice.

repetier, 2015-08-13:
The server does not know about firmware changes if you change eeprom. To change that go to printer configuration and read data from firmware. That should change the settings in configuration. When you then update settings recomputation of all gcodes is started if a change in timing is expected. All gcodes come in a queue. Since timings are computed from these settings only changes to them can change printing time and this is then started automatically.

Update may take a while - all files are reanalysed one after one. You should see it at the modified date of the .gin files in storage/printer/models.

repetier, 2015-08-13:
I here assume you are using 0.60.4 not the version here on github.

miked63017, 2015-08-13:
I am using 0.60.4, but I never saw the times updating, until after I
re-uploaded the g code. Honestly though, I am not sure I gave it enough
time after the printer settings change.

I will mess with it again later today.