repetier/Repetier-Server #74: Moving in wrong way by manual control

orangeudav, 2015-11-07:
My printer is Prusa i3 Rework.
I have Y-endstop at maximum (right) position.
Error reproduction:
1) Press Home Y (now head is at right of printer)
2) On Move Tab select 0.1
3) On Move Tab press Left
4) and my printer trying to move head to RIGHT :( at least 10 or 50, not 0.1
[image]

repetier, 2015-11-08:
Sounds more like wrong configured firmware.

1. left-right is normally the x axis, left is x min.
2. Y axis is front back with extruder over the back part being y max.
3. Positive movement means extruder moves to positive axis. In case of real y it means bed comes out for you.

Since your coordinate system seems wrong or if intentional you did not say how oriented, it is either misunderstanding of right move direction or wrong firmware configuration.

In control tab you can also move sliders and see how bed orientation should be and if you move sliders you see the target coordinate for the axis that gets send to printer.

orangeudav, 2015-11-08:
I not mentioned about interesting detail! When head is at right (maximum) position, first LEFT click try to destroy my printer (move right), but next LEFT click moving head to left as expected.

>> Sounds more like wrong configured firmware.
No, PrintRun (Pronterface) with identicall settings works fine a long time.

>> In control tab you can also move sliders
and sliders have same (wrong) behavior. When slider at right (and head is right after homed), first slide from right to left trying to destroy printer (move head to right after rightest postion)

Haha, i can record video, need it?

repetier, 2015-11-08:
Ok, I get it. I think you have set y home position wrong from picture above. Server uses absolute coordinates and you say y=0 after homing. So after homing host thinks y=0 while you are at y=max. I'm not totally sure what now happens with these confusion. I'm also a bit confused because you mix left/right and y and you x has a x max endstop entered while you say y home endstop.

Please make clear if you are talking about x or y all the time.

You see coordinates in the view. Are these correct before you send the move and does the send move make sense from that point of view, means did it add/subtract to the shown value. Question is here iw we are working with wrong shown values or adding wrong.

orangeudav, 2016-02-10:
Does Repetier-Server have some log with all command sended to printer? I want to see sort of debug log. Gxx, Mxx, etc

repetier, 2016-02-10:
Sure, in the printer select the printer menu icon (gear), there is one called log. There you can say what to log and on next print it will log. In same window you can then later download the log.

orangeudav, 2016-02-10:
Move panel commands not logged into this log :(

repetier, 2016-02-10:
You have to check "log communication while idle", then the communication is logged after the next print/connect into connected.log. After each print the log is overwritten!