repetier/Repetier-Server #76: Feature request: Adjust button sizes!

kyrreaa, 2015-11-21:
We are testing repetier server at the local hackerspace and have run into a major issue.
We use a very large 17" touch monitor as control interface for the printers via repetier server, but the buttons are very hard to hit!
Sometimes we hit delete instead of print!
Other times we hit print instead of view.

repetier, 2015-11-22:
Ok, they be quite close but delete is 15cm away from print, or we are talking about different interfaces.
I have added 10px between print/view/info also it looks ugly now. Will se what the others say about this.

kyrreaa, 2015-11-22:
Isn't there a way to scale parts of the api?
Use 1080x720 resolution and view the interface and assume you are using a mouse cursor about 4x as wide and tall to get an idea of the touchscreen challenge.
Basically the buttons are half as big as your finger.

repetier, 2015-11-22:
Actually I even use it with an iphone for testing sometimes. With the new padding it is no problem to hit them reliable for me.