repetier/Repetier-Server #78: Repetier Server kills Printer

SilentHunter124, 2016-01-17:
When a print job is sent to the printer and in the warm up phase the Repetier Server loses power supply (by e.g. a short circuit) it is not possible to sent any print jobs again to the printer. Only manual control is working but jobs are not being executed. The printer is useless. The only method is, removing the printer and add a new printer (same printer in real) with different name.

repetier, 2016-01-22:
You can print, but the server has 2 modes. If there are no prints started/queued selecting a model starts the print directly.

If you crash server then there is a print in waiting to be printed queue. From that point on you can only start prints from waiting queue. Selecting a model to be printed just adds it to the waiting queue. That is what you have seen.

In host connector there are both queues visible also there is a bug in waiting queue that you can not start it. That will be fixed in next release. As a intermediate solution just delete prints in waiting queue.

SilentHunter124, 2016-01-22:
Yes I understood that, but the problem is, after the reboot (after the server crashed), the queue was no longer visible. It was not possible to delete any jobs...