repetier/Repetier-Server #79: Webcam permission bug

SilentHunter124, 2016-01-17:
If a user has no printing permission amd clicks on webcam, he gets logged out immediately.

repetier, 2016-01-22:
Tried to reproduce the problem but failed. Even a user without permissions could select the webcam tab. Or do you mean an other part with webcam like in control.

Did you use 0.70.1 or an older version?

SilentHunter124, 2016-01-22:
I am using 0.70.1 pro. Yes the user is able to select this tab without permission BUT not when a printing job from an administrator is running. In this case he clicks on the tab webcam and gets logged out immediately.

repetier, 2016-01-25:
I could find out that opening a printer without print permission could lead to a logout, but that had nothing to do with the webcam. Problem here was that the top bar shows switches used for printing and send a command on selection and that caused a permission denied causing the logout. Maybe that also triggers when switching to webcam so I assume for now that it is the same reason and then it will be fixed in next release.

closed 2016-01-25 by repetier