repetier/Repetier-Server #81: Unsafe filename on generation of timelapse video

kyrreaa, 2016-03-11:
Using spaces in filenames for saved jobs on server will not allow the generation of the timelapse video.
Unsure if this is due to lacking hyphens or other issue but for now it's not fun. Images are saved and available though so the video can be generated later if this is fixed.

repetier, 2016-03-12:
On which is did you have this problem?

The filenames are inside a file passed to fmjpeg and i guess there it becomes a problem. Since different os handle arbuments differently it may even be os dependent hence my question.

kyrreaa, 2016-03-12:
I got the message on the git build of ffmpeg using a Raspberry Pi 3.
I think it's the output filename that is the problem, not the inputs as they are in a file as you say.
The output filename is passed on the commandline from what I could tell, without any quotes.

repetier, 2016-03-12:
ok, i will check when im back so next release has it working.