repetier/Repetier-Server #82: Repetier Server possibly causes PC to continuously wake up from Hibernate

puppyscrubber, 2016-03-16:
Hi,
My Win7 PC developed a case of insomnia because it would not stay in Hibernation.
As One might; I spent a good few hours troubleshooting and eventually discovered that Repetier Server either adds some windows timers or waketimers (from a scheduled task?) which where causing this. 

Is this a known issue or unique to my PC?

I recently upgraded Repetier to host V1.6.1 and Repetier Server version is unknown but was likely installed/upgraded with Host V1.6.1.

The windows system event log has an Event id 1 called Power-Troubleshooter which describes why the PC woke from sleep/hibernate however this Event ID reported "Wake Source: Unknown" ... It would obviously be helpful if Windows (or maybe? Rep Server) reported a better reason for the wake from Hibernate! 

Windows command prompt and the "Powercfg -lastwake" and "Powercfg -waketimers" where used to discover the reason for the Insomnia.

Any thoughts?

repetier, 2016-03-16:
Hi,

i have the server on several pcs and none of them wakes from hibernation. But it may prevent going into hibernTion if a job is running. There is a test for this and if it is running it sends a signal not to go to sleep. But i do not even know how to make windows wakeup on its own. You might try stopping the server from the host to see if it is really the server causing this, also i don't know how it should.

Yours
Roland Littwin

apphead, 2016-03-18:
I can confirm the behaviour on Windows 8.1. After stopping the RepetierServer Service the machine did not reboot on suspend / hibernate. 

> i have the server on several pcs and none of them wakes from hibernation. But it may prevent going into hibernTion if a job is running

At the time of testing, no job was running on my W8 machine, BUT I have a RaspberryPi running the server as well and W8 configured as alternative server. Even after shutting down RPB, my W8 Server woke up from sleep and hibernation

Disabling wake timer as described in http://superuser.com/questions/958109/how-to-prevent-windows-10-waking-from-sleep-when-traveling-in-bag
did not work for me (on W8 and with RPB not running).

puppyscrubber, 2016-03-24:
More info from my installation about the wakeups:
1) Uninstalling Rep Server stops the problem
2) Re Installing Repetier host and also selecting server (which downloads server from internet while installing) starts the wakeups again. (I did not uninstall RH, just re installed over the existing RH install)
3) Put PC to Hibernate and after about 1 minute, the PC wakes up and I Log back in at about 09:33 PM
4) Running cmd prompt as admin and issuing "Powercfg -waketimers" gives the below response: (Note the four waketimers set to expire a few minutes after logging back in)
---------------------------------------------
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

C:\Windows\system32>Powercfg -waketimers
Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 09:38:35 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 09:33:53 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 09:33:49 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 09:33:49 PM on 2016/03/24.

  Reason: generic

C:\Windows\system32>
---------------------------------------
*
*
If I run "Powercfg -waketimers" a little later (at 10:05), I get the below response: (New waketimers that have previous and future expiry times so RS is still possibly adding new waketimers
*
*
--------------------------------------
C:\Windows\system32>Powercfg -waketimers
Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 10:15:20 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 10:04:46 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 10:04:44 PM on 2016/03/24.

  Reason: generic

Timer set by [SERVICE] \Device\HarddiskVolume1\Program Files (x86)\Repetier-Serv
er\bin\RepetierServer.exe (RepetierServer) expires at 10:04:44 PM on 2016/03/24.

  Reason: generic
-------------------------------
If I disable waketimers in: Control Panel\Hardware and Sound\Power Options\Edit Plan Settings\change advanced power settings\sleep\allow waketimers ... the auto wake ups obviously stop however this MIGHT disable some R.S. functionality which I cant test as I do not use R.S.

@repetier - fraguada: I am just reporting what I experience with good intentions and THANK YOU for the BEST "HOST" software out there!!!

Rgs, 

Steve

repetier, 2016-03-25:
Strange thing. I could test it now on my W10 virtual machine. Had even more wake up timers. What I do not understand is that I only do

    void RepetierServerApplication::PreventSleepLoop() {
        RLog::log("Prevent sleep thread started.");
        bool stopRequested = false;
        while (!stopRequested)
        {
            try {
                boost::this_thread::sleep(boost::posix_time::milliseconds(5000));
                if(!sleepingAllowed()) {
                    SetThreadExecutionState(ES_SYSTEM_REQUIRED);
                    //RLog::log("Sleep prevented");
                }
                boost::this_thread::interruption_point();
            }
		    catch( boost::thread_interrupted) {
			    stopRequested = true;
			}
			catch(std::exception &ex) {
				RLog::log(ex.what());
			}
		}
        RLog::log("Prevent sleep thread stopped.");
	}

Which should only prevent sleep in 5s interval and does not get called if it only starts. So I'm curious what causes this as I never wrote it to do that, but evidently it does it anyway. Other windows machines do not have the problem which is even more weird. I also think it is not the above function but something else, maybe inside one library I use.

repetier, 2016-03-27:
Ok, I found what sets the wakeuptimer. Some threads get idle and wait with a simple boost::this_thread::sleep function and exactly that functions sets the wakeup timer.

I have now replaced that function for windows with a different sleep function that does not set wakeup timer. So in 0.75 it should be able to go to sleep again.

closed 2016-03-27 by repetier

apphead, 2016-03-27:
Excellent! Thx for fixing the annoyance so quickly!