repetier/Repetier-Server #84: Upload G-Code file programatically

dragondgold, 2016-04-04:
Is it currently possible to upload G-Codes through the API? I see there's the command 'copyModel' but that just adds and already uploaded G-Code to the print queue. Also, manually adding the G-Code file to `ProgramData\Repetier-Server\printer\MCF_Printer\models` doesn't seems to work as Repetier doesn't detects it.

repetier, 2016-04-05:
Sure it is. The frontend does so it must be possible:-)
Here a matching curl command
curl -i -X POST -H 'Content-Type: multipart/form-data' -F 'filename=@" & realFile & "' -F 'a=upload' -F 'apikey=11111111-8393-11e4-a892-00224d7b76cb' 'http://192.168.88.222:3344/printer/model/Eusthatios'

/printer/model = upload
/printer/job = print

next release also has a parameter "group" that defaults to "#" = Default. Has to do with the folders we added in 0.75.

repetier, 2016-04-05:
PS: You can also use "sess" from websocket instead of api-key!

dragondgold, 2016-04-05:
Thank you! Working great! Is it possible to contribute to the API docs? More examples with more details would be really helpful :)

repetier, 2016-04-06:
You know docs are always the latest thing programmers do :-) 

We will update the docs soon to contain some missing or new api calls for 0.75 which hopefully comes in april. I doubt I find the time to make more examples from our side. If you have some great examples we could include I would check them and try to find a good place for them. Eventually even a single example webpage we would link to from main docs.

dragondgold, 2016-04-07:
Haha yeah it's ok :) Great! Looking forward for the new features! :)

closed 2016-04-07 by dragondgold