repetier/Repetier-Server #86: Remember login does not work

apphead, 2016-04-14:
No matter if I use Chrome, FF on PC or Android, the checkbox "Remember me" on the login page does not set a cookie. Since the same functionality works on other sites with these browsers, I assume they are not the cause.
Am I the only one having this issue?

repetier, 2016-04-14:
Remember me expires when you restart the server. Could that be the cause?

apphead, 2016-04-14:
Nope, I hardly ever restart the server (running on a RBP 2). So you don't use cookies then? That would make it persistent across restarts.

repetier, 2016-04-14:
We do not store password/id only a long living session if I remember right. And these are stored in server ram so they are lost on restart.

When you login can you reload several times without getting logged out?

repetier, 2016-04-14:
Which version are you using btw?

apphead, 2016-04-15:
I am using 0.70.1 on. Just found out that there is an upgrade to 0.75. Already installing. Excited!

Personally, I'd prefer to use cookies, but since I can store PW in browser, I will go down that road.

repetier, 2016-04-16:
I have just checked it. We do in deed not use cookies as we need the data only client side. We use local storage instead. Seems like you have disabled local storage then.