repetier/Repetier-Server #87: WS API - setLogLevel

Links2004, 2016-04-24:
I try to get all G code commands and responses from my printer via web sockets.

i connect via:
```
ws://127.0.0.1:4433/socket/?lang=en&apikey=[myapikeyhere]
```

i can send commands to the server and get answers back (I send a ping every 4sec).

but when i send:
```json
{"action":"setLogLevel","printer":"Prusa","callback_id":2,"data":{"level":15}}
````
i get:
```json
{
    "callback_id": 2,
    "data": {},
    "session": "eeysGT4JX41kfeKIVsTjiwjtLl2e5F5v"
}
```

but never see any log events, I tried many levels but see any event.

is there anything else i need to send?
the API list noting else.
http://www.repetier-server.com/manuals/programming/API/index.html

repetier, 2016-04-24:
That is a error in the docs. Should be setLoglevel with lower l in level.

Links2004, 2016-04-24:
confirmed, its now working.
may let the server return a error message/code if the command is unknown/faulty.

closed 2016-04-25 by Links2004