repetier/Repetier-Server #92: Time estimation is completely wrong

sigxcpu76, 2016-09-05:
Hello,

Repetier Server used to be fine in estimating times and out of the blue (I think) it started to show huge times (compared to what slicer is estimating and real wallclock time).

I've attached an example g-code file, estimated at 34 minutes by the slicer and 2h:49m by Repetier Server. I've also attached the printer settings page (auto-detected from EEPROM).

If there is any debug setting that I can enable to check where the time jumps it would be nice.

[belt_support(0).gcode.txt](https://github.com/repetier/Repetier-Server/files/455039/belt_support.0.gcode.txt)

[image]

repetier, 2016-09-05:
Just imported it to my development server and it say 50 minutes, which sounds reasonable at least. So maybe I fixed the issue already or it is setting based. You can view the gcode in server and see which layer takes how long. First and last layer are most interesting since there will be cooling/heating time added and also some command on init/exit lately showed some surprices. When you know which layer it is you should be able to find the code causing the new slowdown. My guess is the either you changed some parameters or kisslicer changed start/end gcode or you did not use kisslicer before so that you now have new commands not present before.

sigxcpu76, 2016-09-05:
In 2D view each layer is shown as ~5 minutes (but it takes much less in real life) so it is not heatup/cooldown related. Being 29 layers it means ~2 hours and 25 minutes which is inline with 2:49 estimated (adding heatup time).

So it is not a specific layer, all layers are over-estimated.

I've uploaded the G-Code here (http://gcode.ws) and it shows 23 minutes, so much less.
I'm using KISlicer for few months now and the "good" estimations were sliced by KISS, too.

In the original post I've attached the printer settings (speeds) and you can see that there is no capping anywhere.

repetier, 2016-09-05:
That is bad. For me it shows around 90s depending on layer. Looking inside a mid layer it is only a list of G1 moves, nothing special so only acceleration/max speed should limit speed and you max speed is much higher. Can you export your printer configuration so I can test with your configuration.

sigxcpu76, 2016-09-05:
I've attached the EEPROM settings (please remove .txt as GitHub is limiting attachment extensions).

[P3 Steel.epr.txt](https://github.com/repetier/Repetier-Server/files/455136/P3.Steel.epr.txt)

repetier, 2016-09-05:
No, not your eeprom that is not important here. I mean the printer configuration of your server. In dashboard you can export the printer settings. These values are used for time estimation.

sigxcpu76, 2016-09-05:
Never used that button :)

Attached to this comment as requested.

[P3_Steel.xml.txt](https://github.com/repetier/Repetier-Server/files/455427/P3_Steel.xml.txt)

repetier, 2016-09-05:
It is very handy. It is a backup of your server configuration and I could now import your settings with one click and also got your strange timings even though my printer config has a better prediction. So I'm curious what is causing the difference here.

sigxcpu76, 2016-09-05:
My prinder (this P3 Steel) config had a better prediction about one month ago. Didn't use it since then and since about one week ago I've noticed strange time estimations. I am not saying that I didn't change anything but I don't remember what I did if I did and what could cause such a dramatic increase.

As far as I know capping the speeds and temperature rise estimations will do this. But as you can see both are pretty OK. Also, I'm not heating the bed through G-Code, as I find it more convenient to do it by manually before starting printing.

sigxcpu76, 2016-09-07:
Any news on this?

repetier, 2016-09-07:
Not really. I noticed you had speed and flow multiplier on 25% so I guess it has something to do with it. but setting it back to 100 did not help. So still need to investigate where it comes from and that is not easy and very time consuming. But at least I could reproduce the error so it is only a question of time.

sigxcpu76, 2016-09-07:
Hm, they are 100% in my UI. Where did you find them at 25%?

[image]

repetier, 2016-09-07:
After importing your setup but that does not seem to be the problem.
Also loaded other gcodes where there was also a difference but not as big as in your example. So there must be some computation that adds a wrong time somehow. Will try splitting it to see what part is causing this.

sigxcpu76, 2016-09-07:
That's strange because I could not find anything regarding 25 in EEPROM dump or config dump. Also the UI is clearly showing 100/110. And I've never had flow at 25% for sure. Speed, maybe, long time ago, to test something. But 25% flow is non-sense :)

repetier, 2016-09-07:
I think it is because it was not connected to a printer. On connection it resets to 100% anyway.

repetier, 2016-09-07:
Ok, I got it. You have set in extruder jerk to 1. This is a very low value for a extruder. Main problem is that server takes this also as max extruder speed (which is a server bug), and since you print faster then 1mm/s extruder speed this limits your speed wrongly. Increase extruder jerk to 10 and times will be computed normal until next release, which will fix this problem.

closed 2016-09-07 by repetier

sigxcpu76, 2016-09-07:
Yes. This fixed it. Thanks!