repetier/Repetier-Server #93: 0.75.1: Communication corruption on native USB!

kyrreaa, 2016-10-06:
Getting some really weird corruption on communications with server 0.75.1 running on Raspberry Pi 3 - with Jesse. (Running armhf version.)
[0.75.1-log.txt](https://github.com/repetier/Repetier-Server/files/513644/0.75.1-log.txt)
Connecting the same cable to pc shows normal communications.

This is the native Arduino Due USB port so the vcp baudrate should not matter, but it is matching the firmware. (It actually doesn't work if not matched!)

I also notice that protocol choice is gone from 0.75.1 ?
Being a stickler for the binary protocol I never had these issues before.

repetier, 2016-10-06:
You never could select protocol in server since quite a while. It was always preset in firmware xml definition file and for repetier defaults to binary.

Log shows the the binary protocol is not the problem, that direction seems to work. Only answers seem not able to have more then 3-4 correct chars in a row. Have you compiled firmware for both ports or native serial exclusively? Have Pi 3 running with programming port with no issues. Would that work on your pi 3 or also cause problems? I mean server does not distinguish between those - absolutely same code. So normally I'd assume the same problem if it comes from server - or there is a difference I currently don't see.

kyrreaa, 2016-10-06:
Interesting.
I have it set to allow use of both ports but it does not work on the prog port.
Not on pc nor on pi.

It is strange however that it does work on pc.
(Albeit I have had 3 hangs where everything just locks up and a reset of pi is not enough and I need to disconnect host, reset pi, and then reconnect host to get the VCP back. Happened on random spots running git master fetched 3 days ago.)

kyrreaa, 2016-10-06:
Btw, with echo on it's hard to see what is coming from where, especially on Repetier Host.
There should possibly be a way to turn on a byte in the log to indicate H or P for host or printer?

I see ju,mps in temp graph to 2000+ btw when it apparantly misinterprets or somehow gets data corrupted. With binary protocol this should never be possible though???

repetier, 2016-10-06:
Binary protocol is only in sending direction, answers are always in ascii. Adding a resend/verify here would drop speed significantly so that is no real option. Only thing would be adding a checksum so host knows if it can trust the answer. Would at least solve the temperature jump problem.

The temp. jumps are errors in temp. reads where a digit was added. Happens if communication is working bad.

One critical thing on Pi 3 is power/voltage. It is very sensitive on voltage so you need AWG20 usb cable and a good power unit. I think the official raspi power delivers more then 5V. If you have a screen connected you can see it on the rainbow that power is critical. This causes often hangs/reset on serial connection and errors.

Regarding log I have to think how many lines would be affected by that problem. Most come from server as it does all the work, only slicer and some other info messages would come from host.

kyrreaa, 2016-10-07:
Got some more data here and it appears it may be a interface problem due to bad tolerance on crystal on the due board.
I will do some accurate measurements later but i did find that the prog port didn't work at 230400 baud without massive corruption. At 250000 it does work but I suspect this is because this rate is a correct multiple so the errors do not compound. I'll measure the actual clock on the sam3x later to verify but so far it appears to be the issue.

What is bad however is the inability over native usb to recover the connection.
This could possibly be related to windows though as it requires unit to reconnect after the program has closed the port, or the port will be busy/locked and unable to re-establish link.

Using the prog port which will always be up regardless of firmware state seems simply put a stabler and safer alternative.  Sad but true.