repetier/Repetier-Server #94: Server ignored Layer 1, went straight to print Layer 2

innovo3d, 2017-05-07:
I don't have a print log for this because I have disabled logs due to some issues I'm having with WiFi, however, I just started a print and it skipped layer 1 and went straight to layer 2.

I've printed this print many times, so I was expecting a brim.  I found it odd that it didn't print the brim, so I checked on Server and saw that it was at about 5% complete, then checked and verified that it was indeed printing layer 2.  

Side note: I was using the heated bed and it takes quite a long time to reach 135 degrees.  I'm not sure if it moved to Z=0.00mm or Z=0.15mm for the second layer.

repetier, 2017-05-07:
Server reads the file line by line and has no option to start somewhere else. So I'm quite sure it started from the beginning. If you started heating extruder too late it might have skipped extrusion since it does not extrude when cold, which might a reason why you are missing layer 1.

innovo3d, 2017-05-08:
I watched the bed move up and it started extruding filament right away, however, it wasn't extruding the brim.  That's when I looked to see that it had started on layer 2.  

I'm 100% positive that it skipped layer 1 from starting the print.  This may or may not have to do with the communication errors that we were talking about.  I think some data is being missed.

repetier, 2017-05-08:
Is this repeatable? Then please enable log to see what is going on. Can currently not imagine how to miss exactly one layer for sending. Only function removing moves is the exclude area function and I do not assume you have excluded anything.