repetier/Repetier-Server #99: Tronxy P802M on Marlin restarts.

rommaster, 2017-11-05:
Hi, my problem is that my Marlin printer can restart while printing from Orange Pi. what could be the problem?

repetier, 2017-11-06:
Enable log to see restart reason. Watchdog would be firmware internal hang, brownout a problem with power and reset can be several reasons. Also check host log var syslog on the pi for possible reasons. Linux closes ports in some conditions and then server will reconnect which resets printer.

rommaster, 2017-11-06:
it says "external reset" in console when printer restarts. i have read that usb 2.0 in linux has some kind of power saver mode, i'll try to disable it. I had similar problem with Octoprint, it just disconnected and didn't reconnect. Printer didn't restart at that time, so the problem could be in usb connection. i'll write you back if it worked.

rommaster, 2017-11-14:
The fault was in USB power management. I didn’t mange to shut down standby mode, but i connected usn relay and printer to a single usb via usb hub and everything started to work good

closed 2017-11-14 by rommaster