repetier/Repetier-Server #100: Problem connecting to printer on armhf version

Krakawet, 2017-11-10:
Hi folks !
I'm struggling to connect a K8200 printer to a rpi3 running repetier-server 0.86.2.
Everything works fine on a PC with the 0.86.2 windows version, but on the Rpi i'm stuck on the second step of the printer setup.
I define my firmware, Marlin, my baudRate at 250000, then I tried every port that seems involved (/dev/serial/by-id/usb-FTDI_FT231X_USB_UART_DA00DDXD-if00-port0, /dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0-port0, /dev/tty/USB0, /dev/tty/AMA0, )
but whatever I may try, I'm stuck on "En attente de connexion", that forbids me to go to step 3, and to see my printer connected.

I tried to use the reset button on the printer board, as unplugging-plugging the USB, with no more luck.
/var/lib/Repetier-Server/logs/server.log don't shown any error.

What else can I do ? Is there other relevant logs anywhere ?

repetier, 2017-11-10:
Did you use our pi image or just install the server?

One reason might be that user repetierserver is not member of group dialout which is required to connect on linux systems. Normally setup adds user to group but had users where this failed and having similar problems.

Krakawet, 2017-11-10:
Thank you Mr. repetier for putting me back on the right track !
It was indeed a permission issue on /dev/ttyUSB0, that I temporary solved with : chmod 777 /dev/ttyUSB0
(this trick didn't survive a reboot)

Strange thing, I tried to fix this permanenlty using : 
```
sudo adduser repetierserver dialout
sudo adduser repetierserver tty
```
 it seems to be effective, as stated in my /etc/group file:
> tty:x:5:repetierserver,pi
> dialout:x:20:pi,repetierserver,
but didn't solve the permission issue.

Anyway, I'm fine with the chmod trick, so we can close this ticket :)

closed 2017-11-11 by Krakawet

repetier, 2017-11-11:
Why adding group pi? Who is the owner of the serial? For me it is dialout

crw-rw----  1 root dialout 166,   0 Nov 11 08:11 ttyACM0