repetier/Repetier-Server #101: Feature request: Resume

badvision, 2018-04-10:
Would it be possible to add auto-resume?  I think it would work something like:

Track the current job
Current line in the file being printed
Current bed/hotend temps
Current Feed speeds/modes (relative vs. absolute, etc)

If the process starts and finds an unfinished job it should offer to continue with a quick start up procedure:
- Re-heat nozzle to previous temp
- Home X
- Home Y
- Move to "safe coordinate" such as X=0, Y=MAX
- Home Z
- Move Z back to last Z height observed
- Resume G Code normally

badvision, 2018-04-10:
Other possibility is to use G92 to set the Z position instead of rehoming Z at all.   Seems that's a generally better-accepted idea

repetier, 2018-04-12:
We are thinking about this in one of next releases. There are of course several problems - this is required on various conditions and printer types. If you do not home to z max homing z is normally not possible with bed being occupied. Without you can not say what z you are especially with autoleveling enabled where Z also depends on x and y. And last xy is never known due to decoupled print buffering on firmware side. Only time we can safely store these informations is when printer disconnects due to some problem. If the pi for example crashes or server we can not write to a file. What might be practical is log every layer or every 60 seconds to a file and flush file (to ensure it is on disc) with all informations. Then we have at least a good starting point.