repetier/Repetier-Server #104: Heater decoupled

leoheck, 2018-08-07:
I was trying to print a model yesterday with no success for 5 tries. Everithing was working just fine last week.

I was printing from the repetier-server (with RPi) and I was keeping the repetier-host connected during the process. The printing stopped on the 2nd layer with "Heater decoupled" issue all the time.

I thought it was something related with the repetier-firmware but I was able to print (a different model after failing 5 times) from the repetier-host (on Ubuntu). The issue was reported here. https://github.com/repetier/Repetier-Firmware/issues/811

So, I think there is an issue printing repetier-server that it generates this "Heater decoupled" or something strange happens when the repetier-host is connected.

repetier, 2018-08-07:
Heater decoupled is always firmware related. Host/server just send the new temperature but firmware checks for decoupling. Therefore closing it on server issue.

closed 2018-08-07 by repetier

leoheck, 2018-08-07:
But my the firmware is working for months printing things from the repetier-host.
This thing happens when I print directly from the server.

repetier, 2018-08-07:
You should have a look at the error message. It says why it decouples and which variable to increase so it does not happen. My guess is that the gcodes differ. Often you trigger this if a fan enables and starts cooling heater and you get the error before heater could correct the extra cooling. But believe me it is the firmware sending the error message and it does it only if your temperature is outside a defined tolerance. This error is always leaving the tolerance. Another example is switching from cooling to heating. This can trigger the error while waiting for 100% cooldown and then reheat will work. Simply because reversing on higher temperatures takes longer then from cold temperatures so the timeout might trigger there and not with the fast start from low temperatures.