repetier/Repetier-Server #107: Announce in the network with Zeroconf

probonopd, 2018-11-03:
Repetier-Server should announce to the network with Zeroconf (Avahi). This would allow one to find Repetier-Server instances on the network easily.

This would be _very_ useful since arguably many Repetier-Server systems are headless and are connected to a network. To access them, the user then normally has to 
* Find out the IP address of the router, access the router as an administrator, and find out the IP address of the device from the list of assigned DHCP addresses (impossible if you have no administrative access to the router); or
* arp-scan the network for the MAC address of the device (impossible if you don't know the MAC address of the device and cumbersome if you do); or
* Walk to the printer and read the IP address from the display

This is how OctoPrint does it:
http://docs.octoprint.org/en/master/bundledplugins/discovery.html

repetier, 2018-11-03:
Our server images here
https://www.repetier-server.com/download-images/
do announce them self using avahi. This just requires avahi to be installed. More exactly the server ip announces it self. So you can use the name.local:3344 regardless of the ip they get.

probonopd, 2018-11-03:
Well, I am not using a Raspberry Pi nor the images linked above. 

I think the application should announce each printer (this is in the scope of the application since only the application knows how many and which printers are attached). The operating system can only announce the machine itself.

repetier, 2018-11-03:
That requires a cross platform library for C++ which I haven't found yet. Everything else would be a pain to handle as each os will then have own different solutions.  Do you know a C++ library not requiring avhci or similar and implement it cross platform?

Normally there is one server per computer so announcing it in avahi as service would be enough. I guess you mean to point each printer to it's gcode start page instead of home screen.

Once I find a library for this I guess I even will do it.

probonopd, 2018-11-03:
Possibly https://opensource.apple.com/tarballs/mDNSResponder/

probonopd, 2018-11-03:
https://github.com/HBPVIS/Servus is a cross-platform library, the only mandatory dependency is a C++11 compiler. Zeroconf will be available in those platforms were either Avahi or DNSSD are available, otherwise an empty dummy backend is used.

probonopd, 2018-11-03:
https://github.com/istopwg/ippsample/blob/52dd8675257f36738a15e56be5182fda4367e517/server/printer.c#L2091-L2115 may also be relevant. Apparently the [IEEE-ISTO Printer Working Group](http://www.pwg.org/) has a standard for how 3D printers should announce themselves on the network.

http://www.pwg.org/3d/