repetier/Repetier-Server #108: Also show port on the display

probonopd, 2018-11-03:
In addition to the IP address, Repetier-Server should also show the port in the display of the attached printer.

This would be very useful especially in situations in which multiple instances of Repetier-Server are running, serving multiple printers.

Personally I have to look up the documentation each time to find out port 3344 ff.

repetier, 2018-11-03:
Why not run all printers form same instance?
Port is a good idea. would at least fir for 20char displays.

repetier, 2018-11-03:
Just see that I also add IP: at start so it will not fit any more. But I guess most will recognice it as an ip number so I can omit it.

repetier, 2018-11-03:
Added for next release.

closed 2018-11-03 by repetier

probonopd, 2018-11-03:
Correction, I meant "if the instance of Repetier-Host has more than one printer attached to it". Thank you very much for adding this.